OpenCyphal/public_regulated_data_types #3: Future: Self-defining data types

pavel-kirienko, 2015-08-18:
The need for this feature is still unclear, but it's worth considering.

An obvious and rather direct approach is to define a message type containing a raw block of bytes with encoded message. The message itself could be encoded in one of the existing self-defining protocol formats, e.g. [Google Protobuf](https://developers.google.com/protocol-buffers/?hl=en), [BSON](https://en.wikipedia.org/wiki/BSON), [etc](https://en.wikipedia.org/wiki/Comparison_of_data_serialization_formats).

Example:

```
void1
uint7 RECIPIENT_BROADCAST = 0
uint7 recipient
Name name
Encoding encoding
uint8[<=1024] payload
```

Where `Name`:

```
void2
uint8[<64] name
```

`Encoding`:
```
uint8 FORMAT_GOOGLE_PROTOBUF = 0
# Other formats...
uint8 format
```

A similar approach can be used for self-defining service calls.

antoinealb, 2015-08-20:
I think ProtoBuf is not self defining, you have statically define
messages, like in UAVCAN. Another good encoding that we use for our
robots is MessagePack (http://msgpack.org).

LorenzMeier, 2015-08-20:
@pavel-kirienko @antoinealb You will want to look at this issue:
https://github.com/mavlink/v2_playground/issues/1

It has a complete table. We're currently revisiting the whole message definition on the vehicle in collaboration with Dronecode & 3DR. There was also some interest to have the same messages on CAN, so it would be a big plus to have native support in UAVCAN for the message format. So far Cap'n Proto seems to lead in terms of features and efficiency.

pavel-kirienko, 2015-10-01:
First stab, at 3DR's request: https://github.com/UAVCAN/dsdl/tree/foreign_object/uavcan/foreign_object

magicrub, 2018-03-13:
@pavel-kirienko I'm interested in reviving this foreign_object concept for carrying multiple protocols over CAN. I see your branch with the generic 1024 byte buffer. I'm surprised FORMATS does not contain MAVLink? I think that's worthwhile. I definitely have a need for it.

antoinealb, 2018-03-14:
You can add it to `Format.uavcan`. However I wonder if there is really interest in putting this kind of support in the standard, as the decoding of the payload will probably be application specific anyway.

magicrub, 2018-03-14:
@antoinealb  I find it useful and it's not just me. Why not include a generic interface like this in the main spec? I don't see a downside. I'll PR the branch from @pavel-kirienko 

Having a bridge is very useful. Another bridges that would be helpful is I2C over CAN. I know ArduPilot would like that. Helps with migrating devices.

pavel-kirienko, 2018-03-22:
Revised implementation at https://github.com/UAVCAN/dsdl/pull/40

closed 2018-03-23 by pavel-kirienko