OpenCyphal/public_regulated_data_types #30: Add an alternative connector option for larger systems

pavel-kirienko, 2017-06-15:
See this thread: <https://groups.google.com/d/msg/uavcan/Er2ghsqumVs/1671vlOtCgAJ>

TL;DR: we're going to add an alternative connector option that will be compatible with the CiA 303 DB9 connector specification (the de-facto industry standard D-Sub connector for CAN).

[image]

The power supply voltage in this pinout is ranging from +18 to +30 VDC, the nominal is +24 VDC. This is unlike the CLIK-Mates, where we're supplying +5 VDC.

This proposition doesn't rule out the possible emergence of other connector types in the future.

Here's the CiA 303 document: [303_1v01080001.pdf](https://github.com/UAVCAN/dsdl/files/1078800/303_1v01080001.pdf)

proficnc, 2017-06-15:
I fully support this.

proficnc, 2017-06-15:
I would like to see an M5 industrial style connection for miniature boards as well.

WickedShell, 2017-06-15:
+1 to a smaller circular connector that still allows us to have cable shielding

LorenzMeier, 2017-06-15:
Yes, that sounds like a really good idea. Should we try to collect a few particular models here? Out of all the alternatives I looked at they seemed the strongest option (among automotive and micro-D).

LorenzMeier, 2017-06-15:
e.g. here is one example:
https://www.digikey.com/product-detail/en/norcomp-inc/852-004-113R001/NOR1374-ND/5278814

and an IP-rated cable with one twisted pair and four pins:
https://www.digikey.com/product-detail/en/norcomp-inc/CCA-000-M02R202/NOR1457-ND/5639518

LorenzMeier, 2017-06-15:
[image]

[image]

WickedShell, 2017-06-15:
Also strong bonus for something that is inherently water proof.

pavel-kirienko, 2017-06-15:
I entirely support the idea of adding a small circular connector type to the specification that would fill the gap between JST GH and DB9. The options Lorenz posted do look well and I could find nothing to complain about on the technical level.

However, can we please talk a bit about the M8 type rather than M5? I do understand the size and weight considerations, but I'm also concerned about the availability of necessary parts and tools. The CiA has a well-defined connector type titled "pico" (what a misnomer) that is based on the M8 type (shown below). Therefore, I expect that M8 type connectors and cables will be easier to source than M5 (which is not used in any major deployments AFAIK). Indeed, a quick googling confirms that there is a large ecosystem out there that we could build upon.

[image]

j3qq4hch, 2017-06-16:
I've put a little picture to compare proposed connectors in a little more obvious way:
[image]

You may notice that DB9 is enormous. 
M8 isn't a smalll thing too.
M5 connectors seem quiet good, tough.
But M5 and M8 aren't  meant to be PCB mount by manufacturer: 
[image]

Can it be a problem?

LorenzMeier, 2017-06-16:
Is the other one really M8 or is it M12?

In terms of PCB mount: That is not an issue. An IP67 setup would require a case with an o-ring and internal wiring. So it might be JST-GH to M5 inside the case.

j3qq4hch, 2017-06-16:
It is M8. Its a beast.
>>An IP67 setup would require a case with an o-ring 

So all the devices are gointg to be IP67 rated?

proficnc, 2017-06-16:
Yes, I think that graphic covers the issue well.

Many M8 connectors are available in PCB mounting.

M8 offers an industry standard. 

If someone could find a standard M5 that is PCB mountable, that would be very nice..

j3qq4hch, 2017-06-16:
But the good news about M8 round connectors is also the fact that they are used even in cheap chinese soldering equipment, so they can be sourced pretty inexpensive

LorenzMeier, 2017-06-16:
I wouldn't say that everything needs to be IP67 rated, but if you spend $$$ on connectors and cables it makes little sense to not protect the PCB with a proper case. If that is all not required you can stay with JST GH which offer basic protection needed for non-sealed systems.

j3qq4hch, 2017-06-16:
I can't find any pcb-mount round connectors at all. All examples of PCB-mount round connectors that I can recall are in fact panel-mount connectors that were mount on PCB or a special adapter-pcb. 
Do they really exist?

pavel-kirienko, 2017-06-16:
> Do they really exist?

There are M12 and M5 board mountable connectors out there, but not M8 that I could find (weird). I don't think this is a deal breaker though; these types of connectors are not for board-level designs.

> An IP67 setup would require a case with an o-ring and internal wiring. So it might be JST-GH to M5 inside the case.

Yes.

> if you spend $$$ on connectors and cables it makes little sense to not protect the PCB with a proper case. If that is all not required you can stay with JST GH which offer basic protection needed for non-sealed systems.

Yes.

-----

Incidentally, I found out that NorComp can't be trusted. Here's what their datasheet says about the pinout of their M8 panel mounted connector:

[image]

And this is what the corresponding STEP model looks like:

[image]

Either way it doesn't match the pin arrangement recommended by the CiA spec, which brings me to the question: does anybody know the formal names for these different pin arrangements? This is what we're after:

[image]

There are at least three different types of pin arrangements, and the vendors tend to be unclear about which ones are used in their connector products.

dimracer, 2017-06-16:
The main advantage of M12, M8, M5 etc connectors is quick and rugged connection. Thats why they ere widely applied in automation for field buses (like CAN ) laying. This type of connectors are assembled on the cables or panels of different automation blocks and devices. I do not think that anybody mount them on boards, because this connector won't be rugged in this case. Usually this type of connector are mounted  on the device panel/case and is connected with board incide it by wires and miniature board connctors. The customer can use cable assemble like this http://mouser.com/ProductDetail/Phoenix-Contact/1525623/?qs=Oehxi%2FWkeHiqd970ABKjFw%3D%3D or make his own cable assembly.

EShamaev, 2017-06-18:
Please take a look at Amphenol PCD Luminus:
https://www.amphenolpcd.com/products/luminus-series

SJS804760 and SJS840410 set is an example.
Datasheets:

[SJS804760_C_REV_3.pdf](https://github.com/UAVCAN/dsdl/files/1083152/SJS804760_C_REV_3.pdf)
[SJS840410_C.pdf](https://github.com/UAVCAN/dsdl/files/1083153/SJS840410_C.pdf)

j3qq4hch, 2017-06-18:
They seem nice. But I can't find any 3d-models of these connectors. Do you know, where to get it?

EShamaev, 2017-06-18:
These are new products from Amphenol, So I guess they will provide 3D models if you ask them for those.

pavel-kirienko, 2017-06-18:
Eugene, Alex, you're way off. What we need is not just a nice connector (there's plenty of them out there), but an industry standard connector. Essentially we're choosing between M5 and M8, and I'm strongly leaning towards the latter due to the current state of affairs with CANopen.

EShamaev, 2017-06-18:
Uhmmm, are we still in aviation sector or moving toward home appliances?

pavel-kirienko, 2017-06-18:
There seem to be nothing wrong with the M series connectors. Many vendors offer hi-rel options in this form factor, too. I am happy to consider other similarly sized industry-standard options as well though.

EShamaev, 2017-06-18:
I consider only MIL spec connectors suitable...
As for M series, I am strongly against soldering wires to pins and if we go crimping - there are barely any options available.

pavel-kirienko, 2017-06-18:
> As for M series, I am strongly against soldering wires to pins and if we go crimping - there are barely any options available.

I agree on the part of avoiding soldered connections. Flying lead terminated M8 and M5 connectors are available even on DigiKey (no soldered wires involved), so that shouldn't really be much of a problem. Consider this, although it is not hi-rel: https://www.digikey.com/product-detail/en/bulgin/PXMBNI08FPM05BFL001/708-2143-ND/6716317. There are PCB mountable options, too.

> I consider only MIL spec connectors suitable...

Why though? You should understand that what we're trying to come up with is a general solution applicable to as many use cases as possible, from low-ish-cost to high reliability, while using commonly available hardware whenever possible. The MIL connector types that I checked right now (shown below) are unreasonably expensive and even more unreasonably hard to source. What exactly are we buying by choosing them? Even if we found compelling arguments to go with MIL series connectors despite the above concerns, are we going to be able to find easily commercially available [cables](http://www.gemac-chemnitz.de/en/products/accessoires/cable-m8-canopen-devicenet-0-6-m-detail)? How about ready-made [termination plugs](https://www.digikey.com/product-detail/en/phoenix-contact/1435988/277-13172-ND/4471670)?

[image]

[image]

-----

On an unrelated note, I found this neat graphic here that demonstrates the differences between different types of pinouts for M series connectors. This answers the question I asked several messages back.

[image]

EShamaev, 2017-06-19:
As for flying lead terminated M8 and M5 connectors I have a concern that these are soldered as well. I could not find any documents stating opposite.

MIL connector refers more to the type of crimping of pins. I should have stated myself more clear as I was meaning MIL Spec contacts in the connector.
For example Amphenol PCD Luminus I proposed is also a MIL type and is not expensive.
Another examples are Harwin high rel connectors that have MIL type contacts and also are not too expensive.

And for general use, I can not say that currently used JST GH is very affordable. The price of the connector itself is low, I agree to that, but price of crimping tool is definitely not for everyone.

proficnc, 2017-06-19:
I think it's quite clear that there are going to be different markets with different needs.

pavel-kirienko, 2017-06-19:
> As for flying lead terminated M8 and M5 connectors I have a concern that these are soldered as well. I could not find any documents stating opposite.

You should also check out the sealed panel-mounted adapter cable assemblies like the one Dmitry has posted above.

Soldered wire connections were used in NASA's safety critical hardware: https://snebulos.mit.edu/projects/reference/NASA-Generic/NASA-STD-8739-3-2.pdf. Although this is an old reference, it is still interesting to know.

> For example Amphenol PCD Luminus I proposed is also a MIL type and is not expensive.
> Another examples are Harwin high rel connectors that have MIL type contacts and also are not too expensive.

I would like to remind here that our objective is to find a suitable standard connector type that is available from more than one vendor. I don't think the connectors you're referring to fit this requirement, so we should stop talking about them.

> And for general use, I can not say that currently used JST GH is very affordable. The price of the connector itself is low, I agree to that, but price of crimping tool is definitely not for everyone.

The properties of JST GH are irrelevant because we're not going to replace it. (By the way, there are [various](http://erlerobotics.com/blog/product/dronecode-cables-and-connectors/) [vendors](https://www.unmannedtechshop.co.uk/dronecode-cables/) who sell DroneCode-compatible JST GH cables and adapters.)

-----

I want to turn this discussion into a more constructive path, so let me summarize the currently known list of the standard connectors here. Unless the list is incomplete, all further discussion should be limited to the listed options.

Type                        | Note                  | Easy to source?   | Compatible CAN standards
----------------------------|-----------------------|-------------------|---------------------------
M5                          | Tiny                  | Yes               | None
M8 B-coded                  | Popular               | Yes               | CANopen, DeviceNet, PROFIBUS, etc.
Micro-D 9-pin               | Twisted pair cables are hard to source | Yes/No (no cables)| None
MIL-C-26482 MS3470L1006PN   | Large (diameter 10 mm)| No                | ARINC 825, CANaerospace
MIL-C-38999 D38999/20FA35PN | Unacceptably huge (diameter 35 mm)| No    | ARINC 825, CANaerospace

EShamaev, 2017-06-19:
Then M8 will be the only option.

pavel-kirienko, 2017-06-19:
[image]

### **<http://doodle.com/poll/m9v7wvckf2caex3c>**

pavel-kirienko, 2017-07-13:
The chapter 8 of the specification has been extended with the new connector options: D-Sub DE-9 and the circular M8. Please familiarize yourself with it here: http://uavcan.org/Specification/8._Hardware_design_recommendations/.

If you have any feedback, please either open a new ticket or submit it to the mailing list.

Thanks!

closed 2017-07-13 by pavel-kirienko