OpenCyphal/public_regulated_data_types #36: [WIP] RFC: Updating data types in a backwards compatible way

kjetilkjeka, 2017-08-16:
Note: This RFC is still highly incomplete. As long as the [WIP] tag is in place I will edit this issue rather freely

# Requirements
- This RFC requires #35 to land, it also requires the use of the briefly mentioned ```@default <VERSION>```.
- This RFC is contradicting to #4, it requires this suggestion to be abandoned.

# Summary

# Explanation

First of all, we need to introduce some concepts.
 - After a data types "implementation period" a fully dsdl compliant sw is required to support reception of the data type.
 - A data types "integration period" starts when the implementation period ends. After the integration period, a fully dsdl compliant sw is allowed to transmit the data type.

kjetilkjeka, 2017-08-17:
@pavel-kirienko I'm close to a finished suggestion here as well.

This ended up beeing much closer to what you propsed in #4 than what I originally suggested. Do you see any lacking bits I should add before removing [WIP] tag?

pavel-kirienko, 2017-08-17:
@kjetilkjeka Sorry about my unresponsiveness, I've been travelling and will continue doing so until tomorrow. I should take a closer look at this and provide a meaningful response around the weekend. Thanks!

kjetilkjeka, 2018-06-25:
Closing this as a superseeding proposal is soon to come

closed 2018-06-25 by kjetilkjeka