OpenCyphal/public_regulated_data_types #103: UAVCAN environement broken

khancyr, 2021-01-28:
Due to the renaming of the repository, all previous works with UAVCAN are now broken as https://github.com/UAVCAN/dsdl doesn't exist anymore.

That wasn't a nice move considering that most of UAVCAN system are on v0 ...

pavel-kirienko, 2021-01-28:
The old name is still valid and can be cloned from. If you have any specific suggestion aside from renaming the repository back, please introduce it.

closed 2021-01-28 by pavel-kirienko

khancyr, 2021-01-28:
That what we got on ArduPilot CI : 
````
 Cloning into '/__w/ardupilot/ardupilot/modules/libcanard/dsdl_compiler/pyuavcan/uavcan/dsdl_files'...
  remote: Not Found
  Error: fatal: repository 'https://github.com/UAVCAN/dsdl.git/' not found
  Fetched in submodule path 'modules/libcanard/dsdl_compiler/pyuavcan/uavcan/dsdl_files', but it did not contain 103745afb42f15f83d49fc23530f7fe2c11358b8. Direct fetching of that commit failed.
  Failed to recurse into submodule path 'modules/libcanard/dsdl_compiler/pyuavcan'
  From git://github.com/ArduPilot/mavlink
   * branch            d8cad47bb3e4ecf21ae631e492e9643cd063cb8c -> FETCH_HEAD
  Submodule path 'modules/mavlink': checked out 'd8cad47bb3e4ecf21ae631e492e9643cd063cb8c'
  Submodule 'pymavlink' (git://github.com/ArduPilot/pymavlink.git) registered for path 'modules/mavlink/pymavlink'
  Cloning into '/__w/ardupilot/ardupilot/modules/mavlink/pymavlink'...
  From git://github.com/ArduPilot/pymavlink
   * branch            1a6faa8b97950158682b932c1444d4488df4adcd -> FETCH_HEAD
  Submodule path 'modules/mavlink/pymavlink': checked out '1a6faa8b97950158682b932c1444d4488df4adcd'
  Submodule path 'modules/uavcan': checked out '3ef4b88d96466bbcf886f74d0ae169a4c8c1bdb0'
  Submodule 'dsdl' (https://github.com/UAVCAN/dsdl) registered for path 'modules/uavcan/dsdl'
  Submodule 'libuavcan/dsdl_compiler/pyuavcan' (https://github.com/UAVCAN/pyuavcan) registered for path 'modules/uavcan/libuavcan/dsdl_compiler/pyuavcan'
  Cloning into '/__w/ardupilot/ardupilot/modules/uavcan/dsdl'...
  Cloning into '/__w/ardupilot/ardupilot/modules/uavcan/libuavcan/dsdl_compiler/pyuavcan'...
  From https://github.com/UAVCAN/dsdl
   * branch            192295c4f9b67f4a20b0eabf74757b6597415f2b -> FETCH_HEAD
  Submodule path 'modules/uavcan/dsdl': checked out '192295c4f9b67f4a20b0eabf74757b6597415f2b'
  From https://github.com/UAVCAN/pyuavcan
   * branch            c58477a644d20ccf95a20c151f3a0402f271c3b8 -> FETCH_HEAD
  Submodule path 'modules/uavcan/libuavcan/dsdl_compiler/pyuavcan': checked out 'c58477a644d20ccf95a20c151f3a0402f271c3b8'
  Submodule 'dsdl' (https://github.com/UAVCAN/dsdl) registered for path 'modules/uavcan/libuavcan/dsdl_compiler/pyuavcan/dsdl'
  Cloning into '/__w/ardupilot/ardupilot/modules/uavcan/libuavcan/dsdl_compiler/pyuavcan/dsdl'...
  From https://github.com/UAVCAN/dsdl
   * branch            fd12483ddd4e58242d61d74a163e7aeaa1e0f466 -> FETCH_HEAD
  Submodule path 'modules/uavcan/libuavcan/dsdl_compiler/pyuavcan/dsdl': checked out 'fd12483ddd4e58242d61d74a163e7aeaa1e0f466'
  Submodule path 'modules/waf': checked out '67b3eac550520cd8a7cd1dce0f77820e91fb8ad0'
  Failed to recurse into submodule path 'modules/libcanard'
  Error: The process '/usr/bin/git' failed with exit code 1
````
So the ability to clone with the old name may just be some DNS cache or something similar ...

pavel-kirienko, 2021-01-28:
Consider updating submodule references to use the new URI. A pull request with that change would be accepted.

khancyr, 2021-01-28:
Fixing it for the master branch isn't an issue. The issue is that all previous release are now broken... And we cannot do anything else that saying to everyone to change to the new url....

Instead of broking compatibility of all previous environment, archiving this repo and create a new one with the new name would have been a better move. And that what Github recommend too