OpenCyphal/public_regulated_data_types #107: used commit is on detached head

BazookaJoe1900, 2021-02-15:
the commit 81526eb6e27095701736dfb6e74842fe07a59f4f is used on PX4 community, but its on non-branch commit (a detached head).
https://github.com/UAVCAN/public_regulated_data_types/commit/81526eb6e27095701736dfb6e74842fe07a59f4f

this is not a good practice, and leads errors on some git clients for some reason.
is it possible to set a branch on that commit?

pavel-kirienko, 2021-02-15:
The commits you are referring to have never been present in this repository.

Their DSDL submodule is outdated and needs to be synced with the upstream. Building any hardware product on top of that is a recipe for disaster.

closed 2021-02-15 by pavel-kirienko

BazookaJoe1900, 2021-02-15:
@pavel-kirienko thanks, good to know, I will notify there