OpenCyphal/public_regulated_data_types #115: Pregenerated headers of public_regulated_data_types to ease development

PetervdPerk-NXP, 2021-04-29:
Would be possible to host the generated headers of public_regulated_data_types on a Github repo? 
This solves some quality of life issues for end-users

- Reduces compilation time of projects using UAVCANv1 since headers are already generated and generating them on the fly with nunavut takes a lot of time.
- Easier initial setup on system without Python, yes there's https://nunaweb.uavcan.org/ but you can't include this into your version control as a submodule
- Easier for a beginner to get starting with UAVCANv1

thirtytwobits, 2021-04-29:
This is akin to checking in binaries so not a great plan. Instead we should fix the performance of Nunavut. Now that Pavel has the pydsdl exponential problem fixed it's on me to fix the jinja problem. This is a priority for me too since I'm about to have two interns suffering with the poor performance.

pavel-kirienko, 2021-04-29:
Scott is right, this is not methodologically correct.

closed 2021-04-29 by pavel-kirienko

thirtytwobits, 2021-04-29:
I will make Nunavut performance a priority @PetervdPerk-NXP . Sorry for your pain.