OpenCyphal/public_regulated_data_types #144: Add a `CMD_IDENTIFY` to ExecuteCommand

Willmac16, 2024-01-28:
For plug and play nodes that support both auto-allocation of NodeIDs and configuration via registers of subject IDs, it would be useful to have an identify command for first time connected nodes (or recently factory reset ones).

If multiple un-configured nodes are connected at the same time/during a power off cycle, then one node at a time could be instructed to identify itself through a light sequence or auditory means, allowing human operators to match auto-allocated nodes IDs to physical nodes. Once identified, the nodes could be configured via registers to publish and subscribe to the appropriate ports, no longer needing identification until the next factory reset.

pavel-kirienko, 2024-01-28:
A pull request adding this command would be accepted.

Willmac16, 2024-01-28:
Should this new constant be added to the deprecated 1.0 version of ExecuteCommand, or only to the current 1.1 version?

pavel-kirienko, 2024-01-28:
Neither. Per section 3.8.3.2 "Versioning principles", a new data type named `ExecuteCommand.1.2` needs to be defined instead, and v1.1 must be marked `@deprecated`.

pavel-kirienko, 2024-01-28:
Done in #145

closed 2024-01-28 by pavel-kirienko